node-modbus/stream #2: please add a socket timeout option

johngeffe, 2017-06-14:
I would like a way to have the socket close automatically.
socket provides two methods already; we just need to pass them into your tcp.js driver.

../drivers/tcp.js file modifications here:

```
// if the timeout option is present; make sure it is an integer
// and limit it between 1 and 60 seconds

if (options.timeout) {
	let timeOut = Math.min(60000, Math.max(1000, parseInt(options.timeout)));
	socket.setTimeout(timeOut);
	socket.on('timeout', () => {
		socket.end();
	});
}

// --- End of timeout option

// if the closeOnData is set to 'true' then
// end() the socket connection when data is returned from
// the tcp connection.

	if (options.closeOnData === true) {
		socket.on('data', () => {
		       socket.end();
	       })
        }
// --- End of closeOnData option
return socket;
```
use like this:
```
modbus.tcp.connect(12345, { timeout: 3000, closeOnData: true, debug: "client" }, (err, connection) => {
    connection.readCoils({ address: 1, quantity: 1 }, (err, res) => {
        if (err) throw err;
        console.log(res.response); // response here
    })
});
```
Thanks!

dresende, 2017-06-14:
This is actually already in place, it might have a unexpected name since I added first for serial connections. Since the transport uses a base object this should work out of the box. Try changing `timeout` to `retry`.

```js
modbus.tcp.connect(12345, { retry: 3000, debug: "client" }, (err, connection) => {
    connection.readCoils({ address: 1, quantity: 1 }, (err, res) => {
        if (err) throw err;
        console.log(res.response); // response here
    })
});
```

As a bonus, you will receive an error of the type `GatewayTargetDeviceFailedToRespond`. 

The default for TCP transport is `30000ms` retry because usually you have a layer that handles packet loss and retransmission, which does not happen on serial (which have a default retry of `500ms`).

dresende, 2017-06-14:
There's also a `retries` option (defaults to `0`) which enables you to tell the transport to try a few more times before returning the `GatewayTargetDeviceFailedToRespond` error.

dresende, 2017-06-20:
Reopen if you find this feature not be working as expected.

closed 2017-06-20 by dresende