node-modbus/stream #4: Recovering from a lost connection

kevingarman, 2017-06-28:
Hi,

I'm probably missing something simple, but how would one detect/cleanup after a connection loss and then reconnect when connectivity is restored?  

Currently I connect and save the connection in a variable.  I use that connection periodically.  If it gets an error (cable unplugged in this case), I call close on the connection.  When the link is restored, I make a new connection and things continue.  However, at some point soon after the reconnect a ECONNRESET is thrown.

What is the proper way to handle connection loss?

Thanks,
Kevin

dresende, 2017-06-28:
I use this module to create a gateway between 2 types of networks, one being modbus. What I do is manage the connection outside and just pass the connection (a stream) to the transport. I'm not sure if this module should handle the connection and reconnect if needed, perhaps just some helper functions that could wrap that functionality.

An `ECONNRESET` in your case, if it's TCP, that's the other endpoint (or a router if in the middle) saying "I don't know anything about this connection, drop it". This can happen quite frequently on mobile networks.

kevingarman, 2017-06-29:
Thanks for taking to time to respond.  I'm sure I'm just missing something...it's probably my lack of proficiency with streams.

I fine with maintaining the connection, but I'm not seeing a way to do that with the connection object returned by `modbus.tcp.connect()`.  If I call `connection.close()`, I would expect the socket/etc to be completely torn down.  But when the network link is restored I get the `ECONNRESET` or sometimes `Socket.writeAfterFIN`...it seems like I shouldn't get these if the connect has been manually closed.

It feels like the modbus library is maybe hanging on to the connection in some way even when I tell it to close.  So I'm not sure how to properly/completely clean up on connection failure so that I can start over when the link is restored.

kevingarman, 2017-06-29:
The following example doesn't exactly match my situation, but I think it illustrates some of what I'm seeing.

```
mb.tcp.connect(502, '10.100.123.220', { debug: 'test', retry: 5000 }, (err, connection) => {
  console.log('connect', err, connection);

  if (!err) {
    connection.readCoils({ address: 1, quantity: 8 }, (err, res) => {
      console.log('readCoils', err, connection);
    });

    connection.close(() => {
      console.log('connection closed manually');
    });
  }
});
```
...and that yields the following output...

```
connect null
Stream {domain: null, _events: Object, _eventsCount: 0, _maxListeners: undefined, transport: Transport, …}
connection closed manually
readCoils null
Stream {domain: null, _events: Object, _eventsCount: 0, _maxListeners: undefined, transport: Transport, …}
Error: read ECONNRESET
    at exports._errnoException (util.js:1016:11)
    at TCP.onread (net.js:609:25)
```
So, it seems like that error should be getting passed back to my readCoils callback so that it can be handled nicely rather than crashing the program.

dresende, 2017-06-29:
I believe you know that your `readCoils` request will not be fulfilled because you're calling `connection.close()` at the same time (connection is a stream instance which asks the transport to close which in turn asks the real socket connection to close).

You should not be receiving the `ECONNRESET` that way. The error is being thrown because you're not listening for the `error` event in connection. But don't try it now, I already tested locally and I'm not propagating the `error` event to the stream so that will be useless.

I'm going to make 2 changes:

1. Being able to listen to `error` event on your `connection` so you can catch errors, log them and do something about it (instead of the process crashing);
2. Check if your `readCoils` also doesn't throw and, instead, return a nice error indicating the connection has gone sometime before.

Thank you for insisting on your explanation 👍

kevingarman, 2017-06-29:
Ya, putting the close there was just to force the error.  Thanks for digging into it.

dresende, 2017-06-29:
Can you test the `master` branch to see if it still happens? You should be able to listen for the `error` in your `connection` object and also receive a fake `ECLOSED` error when calling the `readCoils`.

dresende, 2017-06-29:
I believe my answer was not clear. For your example, the `ECLOSED` error should come up inside your `readCoils`.

For other unexpected errors occurring on other occasions, you should be able to listen for the `error` event.

kevingarman, 2017-06-29:
I'm not seeing the `ECLOSED` and am still getting the `ECONNRESET`.  However, if I add the error handler on the connection, I can intercept it there and keep things alive.

dresende, 2017-06-29:
I tested with something similar to the example you gave. Are you now testing with your real code?

dresende, 2017-06-30:
At least what I did is now on 2e7c1c4f98dbc886c2077eec9d988f02234a1223

kevingarman, 2017-06-30:
I'm using the same test code...

```
mb.tcp.connect(502, '10.100.123.220', { debug: 'test', retry: 5000 }, (err, connection) => {
  console.log('connect', err, connection);

  if (!err) {
    connection.readCoils({ address: 1, quantity: 8 }, (err, res) => {
      console.log('readCoils', err, connection);
    });

    connection.close(() => {
      console.log('connection closed manually');
    });
  }
});
```
...which yields...

```
connect null
Stream {domain: null, _events: Object, _eventsCount: 0, _maxListeners: undefined, transport: Transport, …}
connection closed manually
readCoils null
Stream {domain: null, _events: Object, _eventsCount: 0, _maxListeners: undefined, transport: Transport, …}
Error: read ECONNRESET
    at exports._errnoException (util.js:1016:11)
    at TCP.onread (net.js:609:25)
```

With the addition of an error handler on connection, I have...

```
mb.tcp.connect(502, '10.100.123.220', { debug: 'test', retry: 5000 }, (err, connection) => {
  console.log('connect', err, connection);

  connection.on('error', (err) => {
    console.error('connect', err);
  });

  if (!err) {
    connection.readCoils({ address: 1, quantity: 8 }, (err, res) => {
      console.log('readCoils', err, connection);
    });

    connection.close(() => {
      console.log('connection closed manually');
    });
  }
});
```

...which yields (without crashing)...

```
connect null
Stream {domain: null, _events: Object, _eventsCount: 0, _maxListeners: undefined, transport: Transport, …}
connection closed manually
readCoils null
Stream {domain: null, _events: Object, _eventsCount: 1, _maxListeners: undefined, transport: Transport, …}
connect
Error: read ECONNRESET
```

dresende, 2017-07-03:
What do you have on that ip/port? Do you have the code or is that out of the scope?

dresende, 2017-07-03:
I still can't get your `ECONNRESET` but I now get a `writeAfterFIN` error using this sample.

```js
var modbus = require("modbus-stream");

modbus.tcp.server({ debug: "server" }, function (connection) {
	connection.on("read-coils", function (req, reply) {
		setTimeout(function () {
			return reply(null, [ true ]);
		}, 500);
	});
}).listen(12345, function () {
	console.log("server ready");
	modbus.tcp.connect(12345, { debug: "client" }, function (err, connection) {
		connection.readCoils({ address: 0, quantity: 1 }, function (err, info) {
			if (err) {
				return console.log("readCoils error", err);
			}

			console.log("response", info.response);
		});
		connection.close(function () {
			console.log("connection closed");
		});
	});
});
```

I get this:

```
server ready
2017-07-03T09:59:52.890Z << client 0x[ 00, 01, 00, 00, 00, 06, 01, 01, 00, 00, 00, 01 ]
connection closed
2017-07-03T09:59:52.896Z >> server 0x[ 00, 01, 00, 00, 00, 06, 01, 01, 00, 00, 00, 01 ]
2017-07-03T09:59:53.400Z << server 0x[ 00, 01, 00, 00, 00, 04, 01, 01, 01, 01 ]

Error: This socket has been ended by the other party
    at Socket.writeAfterFIN [as write] (net.js:275:12)
    at Transport.BaseTransport.write (/Users/dresende/Projects/modbus-stream/lib/transport/transport.js:45:15)
    at /Users/dresende/Projects/modbus-stream/lib/transport/transport.js:148:17
    at null._onTimeout (/Users/dresende/Projects/modbus-stream/test.js:10:11)
    at Timer.listOnTimeout [as ontimeout] (timers.js:110:15)
```

I'm going to dig a bit further.

dresende, 2017-07-03:
Forget it, this is a server error, not the client. You can avoid the `throw` by just:

```js
connection.on("error", function (err) {
	console.log("server error", err);
});
```

kevingarman, 2017-07-03:
Works for me :)

dresende, 2017-07-03:
So, it seems your server might be closing the connection right after `readCoils`. I'm now remembering that perhaps your server doesn't allow the connection to be kept alive.

I'm going to simulate that and see if the module handles it nicely. Perhaps a close/end event would be better, although if the server just resets the connection you still need to check for the error event.

dresende, 2017-07-03:
I exposed a `close` event in the new version (`0.26.0`) so you can detect the connection being closed before trying to send something again (and receiving an `error`). I hope this works out for you. If you feel there's still something wrong please reopen the ticket.

closed 2017-07-03 by dresende

kevingarman, 2017-07-03:
Sounds great, thanks for your work!