node-modbus/stream #6: Callback called a second time from "readInputRegisters"

jhillacre, 2017-07-19:
Running into an issue migrating over from `node-modbus-tcp`. I seem to have a `readInputRegisters` callback being called a second time. The first time has my results, the second time has a `GatewayTargetDeviceFailedToRespond` error object.

I'm using the async library to handle callbacks to run multiple `readInputRegisters` in parallel. Async throws an exception when its callback is called a second time. I added a call counter to my function and it seems like the first `readInputRegisters` that returned successfully was then called again.

Is this caused by `transport.retryTimer` being overwritten in [`lib/transport/transport.js`](https://github.com/node-modbus/stream/blob/master/lib/transport/transport.js#L88) by the second call to `readInputRegisters`?

Here is my code:
<details>
  <summary>setup</summary>

```javascript
const modbus = require('modbus-stream');
let modbusConn;
...
modbus.tcp.connect(
  sql.currentSettings.modbusPort,
  sql.currentSettings.ipAddress,
  (err, connection) => {
    if (err) {
      return die(err, 'modbus');
    }
    modbusConn = connection;
  }
);
```

</details>

<details>
  <summary>polled function (called every ~50 ms)</summary>

```javascript
let callCount = 0;
...
function getPlcDMAndHSCPV(cb) {
  let myCallCount = callCount++;
  console.log('myCallCount', myCallCount);
  async.parallel(
    {
      counter: innercb => modbusConn.readInputRegisters(
        {address: 520, quantity: 2},
        (err, result) => {
          console.log('counterCb', {myCallCount: myCallCount, err: err, result: result}); innercb(err, result);
        }
      ),
      voltageAndRanges: innercb => modbusConn.readInputRegisters(
        {address: 1000, quantity: 6},
        (err, result) => {
          console.log('voltageAndRangesCb', {myCallCount: myCallCount, err: err, result: result}); innercb(err, result);
        }
      ),
    },
    (err, results) => {
      if (err) {
        return cb(err);
      }
      let mapped = [];
      mapped[0] = results.voltageAndRanges.response.data[0].readInt16BE(0);
      mapped[1] = results.voltageAndRanges.response.data[1].readInt16BE(0);
      mapped[3] = Buffer.concat(results.counter.response.data).readInt32BE(0);
      mapped[2] = (mapped[3] / pulseToDistanceRatio).toFixed(2);
      mapped[4] = results.voltageAndRanges.response.data[4].readInt16BE(0);
      mapped[5] = results.voltageAndRanges.response.data[5].readInt16BE(0);
      cb(null, mapped);
    }
  );
}
```

</details>

<details>
  <summary>output (counterCb with myCallCount of 0 is called first and last.)</summary>

```
myCallCount 0
counterCb { myCallCount: 0,
  err: null,
  result: 
   { transactionId: 1,
     protocol: 0,
     length: 7,
     unitId: 1,
     pdu: <Buffer 04 04 02 7d d0 27>,
     callback: [Function],
     response: { code: 'ReadInputRegisters', data: [Object] } } }
voltageAndRangesCb { myCallCount: 0,
  err: null,
  result: 
   { transactionId: 2,
     protocol: 0,
     length: 15,
     unitId: 1,
     pdu: <Buffer 04 0c 00 34 00 37 ff ff ff ff 00 01 00 01>,
     callback: [Function],
     response: { code: 'ReadInputRegisters', data: [Object] } } }
```
*... 30 seconds of similar calls ...*

```
myCallCount 599
counterCb { myCallCount: 599,
  err: null,
  result: 
   { transactionId: 1199,
     protocol: 0,
     length: 7,
     unitId: 1,
     pdu: <Buffer 04 04 02 7d b7 aa>,
     callback: [Function],
     response: { code: 'ReadInputRegisters', data: [Object] } } }
voltageAndRangesCb { myCallCount: 599,
  err: null,
  result: 
   { transactionId: 1200,
     protocol: 0,
     length: 15,
     unitId: 1,
     pdu: <Buffer 04 0c 00 3a 00 33 ff ff ff ff 00 01 00 01>,
     callback: [Function],
     response: { code: 'ReadInputRegisters', data: [Object] } } }
counterCb { myCallCount: 0,
  err: 
   { Error: GatewayTargetDeviceFailedToRespond
       at Object.exports.error (/home/joel/WebstormProjects/well-logger/server/node_modules/modbus-pdu/lib/Exception.js:24:13)
       at Timeout._onTimeout (/home/joel/WebstormProjects/well-logger/server/node_modules/modbus-stream/lib/transport/transport.js:89:33)
       at ontimeout (timers.js:386:14)
       at tryOnTimeout (timers.js:250:5)
       at Timer.listOnTimeout (timers.js:214:5) code: 11 },
  result: undefined }
```

</details>

<details>
  <summary>exception (after output)</summary>

```
error: uncaughtException: Callback was already called. 
{ date: 'Wed Jul 19 2017 16:29:20 GMT-0600 (MDT)',
  process: 
   { pid: 15041,
     uid: 1002,
     gid: 1002,
     cwd: '/home/joel/WebstormProjects/well-logger/server',
     execPath: '/home/joel/.nvm/versions/node/v6.11.1/bin/node',
     version: 'v6.11.1',
     argv: 
      [ '/home/joel/.nvm/versions/node/v6.11.1/bin/node',
        '/home/joel/WebstormProjects/well-logger/server/app.js' ],
     memoryUsage: 
      { rss: 57839616,
        heapTotal: 36737024,
        heapUsed: 32882016,
        external: 396488 } },
  os: 
   { loadavg: [ 0.8125, 0.5537109375, 0.38232421875 ],
     uptime: 194414 },
  trace: 
   [ { column: 32,
       file: '/home/joel/WebstormProjects/well-logger/server/node_modules/async/dist/async.js',
       function: null,
       line: 903,
       method: null,
       native: false },
     { column: 13,
       file: '/home/joel/WebstormProjects/well-logger/server/node_modules/async/dist/async.js',
       function: null,
       line: 3835,
       method: null,
       native: false },
     { column: 93,
       file: '/home/joel/WebstormProjects/well-logger/server/lib/plc.js',
       function: 'modbusConn.readInputRegisters',
       line: 108,
       method: 'readInputRegisters',
       native: false },
     { column: 14,
       file: '/home/joel/WebstormProjects/well-logger/server/node_modules/modbus-stream/lib/transport/transport.js',
       function: 'Timeout._onTimeout',
       line: 89,
       method: '_onTimeout',
       native: false },
     { column: 14,
       file: 'timers.js',
       function: 'ontimeout',
       line: 386,
       method: null,
       native: false },
     { column: 5,
       file: 'timers.js',
       function: 'tryOnTimeout',
       line: 250,
       method: null,
       native: false },
     { column: 5,
       file: 'timers.js',
       function: 'Timer.listOnTimeout',
       line: 214,
       method: 'listOnTimeout',
       native: false } ],
  stack: 
   [ 'Error: Callback was already called.',
     '    at /home/joel/WebstormProjects/well-logger/server/node_modules/async/dist/async.js:903:32',
     '    at /home/joel/WebstormProjects/well-logger/server/node_modules/async/dist/async.js:3835:13',
     '    at modbusConn.readInputRegisters (/home/joel/WebstormProjects/well-logger/server/lib/plc.js:108:93)',
     '    at Timeout._onTimeout (/home/joel/WebstormProjects/well-logger/server/node_modules/modbus-stream/lib/transport/transport.js:89:14)',
     '    at ontimeout (timers.js:386:14)',
     '    at tryOnTimeout (timers.js:250:5)',
     '    at Timer.listOnTimeout (timers.js:214:5)' ] }
```

</details>

dresende, 2017-07-20:
Hmm... it's probably because of that. I have a test to check if the callback can be triggered a second time but it just tests for a duplicate package, not the parallelism you refer. I'm going to change the test and fix that. Nice catch!

dresende, 2017-07-20:
What's actually weird is that any valid package will clear the timer, so it's more probable that your callback is never called (because of mixed packages) than a duplicate call.

dresende, 2017-07-20:
Can you please test master and see if this solves? I can't seem to replicate your specific issue but I changed tcp transport to allow multiple timers.

jhillacre, 2017-07-20:
Using master, I'm no longer getting complains about recalled callbacks. Thanks for the quick fix!

dresende, 2017-07-20:
Thank you for testing. I'll create a new version later today.

closed 2017-07-20 by dresende