node-modbus/stream #7: TCP Transport: Transaction ID overflow

jhillacre, 2017-07-20:
Managed to overflow the transaction id counter in [lib/transport/tcp.js](https://github.com/node-modbus/stream/blob/master/lib/transport/tcp.js#L38). I'm not 100% familiar with the raw protocol, but would it be safe for node-modbus-stream to be resetting the transaction ID instead of overflowing? If not, I guess I could catch this and rebuild the connection.

My exception:
```
TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1041:11)
    at Buffer.writeUInt16BE (buffer.js:1110:5)
    at Transport.wrap (/home/logger/server/node_modules/modbus-stream/lib/transport/tcp.js:38:8)
    at Transport.BaseTransport.send (/home/logger/server/node_modules/modbus-stream/lib/transport/transport.js:72:23)
    at Stream.readInputRegisters (/home/logger/server/node_modules/modbus-stream/lib/stream.js:103:17)
    at voltageAndRanges (/home/logger/server/lib/plc.js:104:49)
    at /home/logger/server/node_modules/async/dist/async.js:3853:24
    at replenish (/home/logger/server/node_modules/async/dist/async.js:946:17)
    at /home/logger/server/node_modules/async/dist/async.js:950:9
    at eachOfLimit (/home/logger/server/node_modules/async/dist/async.js:975:24)
    at /home/logger/server/node_modules/async/dist/async.js:980:16
    at eachOf (/home/logger/server/node_modules/async/dist/async.js:1051:5)
```

dresende, 2017-07-21:
I think it's better to reset it. I have to look at it. Can you provide me with the node version you're using? I usually test with 0.10 to ensure it works on arm devices.

closed 2017-07-21 by dresende

jhillacre, 2017-07-21:
> Closed #7 via 5ef6d2a.

Using v6.11.1 on a raspberry pi. (armv7)

Joel

dresende, 2017-07-21:
I already tried on v0.10.29 on a linux from scratch with armel (v5). Slow as hell.

commit 5ef6d2a4 (2017-07-21): tcp: fixes #7