node-modbus/stream #8: uintId option do not work  as expect

JasLin, 2017-09-12:
```
const modbus = require(".")
modbus.tcp.connect(10000, "127.0.0.1", { unitId: 3, debug: "automaton-2454" }, (err, connection) => {
    // do something with connection
    connection.readCoils( {address: 0, quantity:4}, (err, res) => {
        console.log(err)
        console.log(res)
    })
});

```

debug log:

```
2017-09-12T10:23:56.833Z << automaton-2454 0x[ 00, 01, 00, 00, 00, 06, 01, 01, 00, 00, 00, 04 ]

```
unitId still 01, ([ 00, 01, 00, 00, 00, 06, 01(should be 03), 01, 00, 00, 00, 04 ])

dresende, 2017-09-15:
Well, it's not a bug, it's actually a new feature. The constructor does not accept `unitId` there, just an initial `transactionId`, but I'm going to add one.

For your example to work now you could try this:

```js
const modbus = require(".")
modbus.tcp.connect(10000, "127.0.0.1", { debug: "automaton-2454" }, (err, connection) => {
    // do something with connection
    connection.readCoils({ address: 0, quantity: 4, extra: { unitId: 3 }}, (err, res) => {
        console.log(err)
        console.log(res)
    })
});
```

dresende, 2017-09-15:
Checkout version `0.37.0`

closed 2017-09-15 by dresende

JasLin, 2017-09-17:
thanks,  I will try this version later :)

commit 5931ba89 (2017-09-15): tcp: allow default unitId to be passed in constructor (fixes #8)