node-modbus/stream #10: blocks[i].copy is not a function

IlhanSeven, 2017-09-18:
```
var modbus = require("modbus-stream");

module.exports = {
    initModbus: function() {
         modbus.tcp.server({ debug: "server" }, (connection) => {
              connection.readHoldingRegisters( {address: 0, quantity: 1}, (err, info) => {
                   console.log("response", info.response.data);
              });
         }).listen(12345, () => {
              modbus.tcp.connect(12345, { debug: "client" }, (err, connection) => {
                   connection.on("read-holding-registers", (request, reply) => {
                        reply(null, [234]);
                   });
              });
         });
    }
}
```

Hi, thanks for your effort. Could not achieve just a simple thing. What can be wrong here?

Error output:

```
events.js:141
      throw er; // Unhandled 'error' event
      ^

TypeError: blocks[i].copy is not a function
    at Object.blocksToBuffer [as buildResponse] (/home/ubuntu/workspace/EasygenSlave/node_modules/modbus-stream/node_modules/modbus-pdu/lib/Helpers.js:111:13)
    at Object.build (/home/ubuntu/workspace/EasygenSlave/node_modules/modbus-stream/node_modules/modbus-pdu/lib/Modbus.js:113:30)
    at /home/ubuntu/workspace/EasygenSlave/node_modules/modbus-stream/lib/transport/transport.js:165:75
    at Stream.<anonymous> (/home/ubuntu/workspace/EasygenSlave/modbus-tcp-server.js:12:13)
    at emitTwo (events.js:87:13)
    at Stream.emit (events.js:172:7)
    at Transport.<anonymous> (/home/ubuntu/workspace/EasygenSlave/node_modules/modbus-stream/lib/stream.js:44:10)
    at emitThree (events.js:97:13)
    at Transport.emit (events.js:175:7)
    at handle (/home/ubuntu/workspace/EasygenSlave/node_modules/modbus-stream/lib/transport/transport.js:156:16)
```

Can you give me guidance when you are available? Cheers.

dresende, 2017-09-18:
Change the line:

```js
reply(null, [234]);
```

To something like:

```js
reply(null, [ Buffer.from([ 0x02, 0x34]) ]);
```

IlhanSeven, 2017-09-18:
It worked but one thing is not clear to me. What are the addresses for these values? How can we specify them?

dresende, 2017-09-18:
I'm not sure you know the basics around modbus.

In your example you requested address `0` and quantity `1`, so you're asking only for data in address `0`. The reading is sequential so if you would request quantity `3` you would read addresses `0-2`.

In this case, holding registers, each address containers 2 bytes, so I represent those 2 bytes as a nodejs `Buffer` with 2 bytes. If something would request you to read from address `0` to `2`, you could reply something like:

```js
// address 0 = 0x0102
// address 1 = 0x0304
// address 2 = 0x0506
reply(null, [ Buffer.from([ 0x01, 0x02 ]), Buffer.from([ 0x03, 0x04 ]), Buffer.from([ 0x05, 0x06 ]) ]);
```

IlhanSeven, 2017-09-18:
Thanks for the answer. What if master asks for register address 450010? Is there a solution to this?

dresende, 2017-09-18:
If you mean `450 010` that doesn't exist as the addresses are 2 bytes (65 536 addresses).

If you mean `45 010`, you're probably talking about the addresses of the holding registers. The standard does not mention any offset but many hardware suppliers have the datasheets for the holding registers starting at `40 001` which means address `0`. Using this logic, the address `45 010` would be `5 009` (`45 010 - 40 001`).

dresende, 2017-09-18:
http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

dresende, 2017-09-18:
Also, this could help understand the protocol bytes behind the request.

http://www.simplymodbus.ca/FC03.htm

closed 2017-09-18 by dresende

IlhanSeven, 2017-09-19:
Sorry it was (4)50010 which means holding register #50010. Thanks for your help. How can I check if the request is made to this address?

dresende, 2017-09-19:
You can enable debug and look at the bytes in hexadecimal. You have to know the protocol (the site above can help). Or... you can make a real request to a modbus device and see if the responde makes sense.

IlhanSeven, 2017-09-19:
Really sorry, asked too many questions. I meant when we are the slave/server.

dresende, 2017-09-19:
```js
const modbus = require("modbus-stream");

modbus.tcp.server({ debug: "server" }, (connection) => {
	connection.on("error", (err) => {
		console.log("server error", err);
	});
	connection.on("read-coils", (req, reply) => {
		console.log(req); // <--- look at this object
		return reply(null, [ 1, 0, 1, 0, 0, .... ]);
	});
}).listen(502, () => {
	console.log("server ready");
});
```

dresende, 2017-09-19:
As with `read-coils`, you can listen for the other events (`read-holding-registers`, ..).

IlhanSeven, 2017-09-19:
Can we be slave on serial port with this?

dresende, 2017-09-19:
With this module you can create any role and make requests from either role (slave or master)

IlhanSeven, 2017-09-20:
Hi man. Just realized again this is a wonderful work. Pure and easy to use. Is there a ready way to give illegal data exception response?

dresende, 2017-09-20:
Yes, just use the `reply` method and pass a string:

```js
reply("IllegalDataAddress");
```

You can use any of these exception strings [here](https://github.com/node-modbus/pdu#exceptions).

IlhanSeven, 2017-09-20:
Useful as always. Thanks.