node-modbus/stream #11: Disabling logs

IlhanSeven, 2017-10-06:
Hi, 

Is there a way to disable logging. Log files go crazy.

Thanks.

dresende, 2017-10-06:
Try adding option `debug : null`. I should change it to also `undefined`..

IlhanSeven, 2017-10-06:
Thank you. Sorry I must be blind.

closed 2017-10-06 by IlhanSeven

IlhanSeven, 2017-10-06:
Sorry, where should I add it when using modbus serial?

dresende, 2017-10-06:
Like in the [README](https://github.com/node-modbus/stream#connection), where you see `"automaton-123"`, use `null` instead.

IlhanSeven, 2017-10-10:
Hey man, 

Is there an example for writeSingleRegister or writeMultipleRegister or both? Sorry I could not find it in your docs.

dresende, 2017-10-10:
Check [Requests Section](https://github.com/node-modbus/stream#requests).
```js
connection.writeSingleRegister({ address: 123, value: Buffer.from([ 0x12, 0x34 ]) }, (err, info) => {
    console.log("response", info.response);
});
```

IlhanSeven, 2017-10-10:
I was actually using below code for that being a tcp server.

```
connection.on("write-single-register", (req, reply) => {
      registerMap[req.request.address] = req.request.value;
      reply(???)
});
```
Using Modbus Poll I can write a value but I get timeout error in response. What should be in place of question marks in the reply?

dresende, 2017-10-10:
On a reply, try this:

```js
// error = null, value = buffer
reply(null, Buffer.from([ 0x12, 0x34 ]));
```

dresende, 2017-10-10:
You can also trigger standard errors like:

```js
reply("IllegalDataAddress");
```

dresende, 2017-10-10:
https://github.com/node-modbus/pdu#exceptions

IlhanSeven, 2017-10-10:
I tried error = null and echoing back the value written but still getting Timeout Error on Modbus Poll. On the slave side, I am getting:

```
server error TypeError: Cannot read property 'copy' of undefined
    at Object.exports.buildRequest (C:\Users\Ilhan\Desktop\NodeDocker\mixergy-modbus-slave\node_modules\modbus-pdu\lib\protocol\WRIT
    at Object.build (C:\Users\Ilhan\Desktop\NodeDocker\mixergy-modbus-slave\node_modules\modbus-pdu\lib\Modbus.js:113:30)
    at C:\Users\Ilhan\Desktop\NodeDocker\mixergy-modbus-slave\node_modules\modbus-stream\lib\transport\transport.js:165:75
    at Stream.connection.on (C:\Users\Ilhan\Desktop\NodeDocker\mixergy-modbus-slave\modbus-tcp-server.js:34:21)
    at emitTwo (events.js:106:13)
    at Stream.emit (events.js:191:7)
    at Transport.<anonymous> (C:\Users\Ilhan\Desktop\NodeDocker\mixergy-modbus-slave\node_modules\modbus-stream\lib\stream.js:44:10)
    at emitThree (events.js:116:13)
    at Transport.emit (events.js:194:7)
    at handle (C:\Users\Ilhan\Desktop\NodeDocker\mixergy-modbus-slave\node_modules\modbus-stream\lib\transport\transport.js:156:16)
```

Thanks for help man.

dresende, 2017-10-10:
Can you show your code. That error is an indication that it's trying to return a buffer and you didn't supply it. It didn't see an error.

IlhanSeven, 2017-10-10:
Here it is:

```
module.exports = {
    initModbus: function() {
        
        modbus.tcp.server({ debug: null }, (connection) => {
	        connection.on("error", (err) => {
		        console.log("server error", err);
	        });
	
	        connection.on("read-holding-registers", (req, reply) => {
                
                if(req.request.address >= 2000 && req.request.address <= 2299 && (req.request.address + req. request.quantity) <= 2299) {
                    
                    var startIndex = req.request.address - 2000;
                    var endIndex = startIndex + req.request.quantity;
                    var response = new Array();
                    for(var i = startIndex; i < endIndex; i++) {
                        response.push(mixergyMap[i]);
                    }
                    reply(null, response);
                } else {
                    
                    reply("IllegalDataAddress");
                }
            });
            
            connection.on("write-single-register", (req, reply) => {
               
                if(req.request.address >= 2000 && req.request.address <= 2299) {
                    
                    mixergyMap[req.request.address - 2000] = req.request.value;
                    reply(null, mixergyMap[req.request.address-2000]);
                } else {
                    reply("IllegalDataAddress");
                }
            });
            
        }).listen(5502, () => {
	        console.log("server ready");
        });
    }
}
```

IlhanSeven, 2017-10-10:
I think the issue is the reply function. I am not sure if I am using it correctly. According to Modbus standard, we should echo back the request when the function is 'write' (Correct me if I am wrong). I could not figure out how to do this.

dresende, 2017-10-10:
My bad, I forgot something. You are correct, you should echo back the request value **and** the address. You should reply like this:

```js
reply(null, req.request.address, req.request.value);
```