node-modbus/stream #12: "GatewayTargetDeviceFailedToRespond" with multiple "readHoldingRegisters"

insideGen, 2017-10-21:
Hi!
I get this error `GatewayTargetDeviceFailedToRespond` when I execute several times the `readHoldingRegisters` function. I use the `readHoldingRegisters` function inside a loop.
Any idea?
Thanks.

dresende, 2017-10-23:
I never used this synchronous way, you should use `async` module to do this. But I'll try to reproduce it locally and see what's happening.

dresende, 2017-10-23:
Is the server an automaton or is it in node too? If it is, please provide a sample of the code that responds to this request.

dresende, 2017-10-23:
Here's a code that works 100% of the time.

```js
var modbus    = require("modbus-stream");
var startDate = Date.now();
var addresses = 100;

modbus.tcp.server({ debug: null }, function (connection) {
	connection.on("read-holding-registers", function (req, reply) {
		return reply(null, [ Buffer.from([ 0x00, req.request.address ]) ]);
	});
}).listen(12345, function () {
	modbus.tcp.connect(12345, { debug: null }, function (err, connection) {
		var responses = 0;

		for (let i = 0; i < addresses; i++) {
			(function (address) {
				connection.readHoldingRegisters({ address: address, quantity: 1 }, function (err, res) {
					if (err != null) {
						console.log((Date.now() - startDate) + "ms - Register[" + address + "]: " + err.message);
					} else if (res != null) {
						console.log((Date.now() - startDate) + "ms - Register[" + address + "]: " + res.response.data[0].readInt16BE(0));

						if (++responses === addresses) {
							console.log((Date.now() - startDate) + "ms - Successfully received " + responses + " responses");

							process.exit(0);
						}
					}
				});
			})(i);
		}
	});
});
```

Perhaps your modbus server is not replying to every request and you should make a retry. Default tcp retry time is 30 seconds with no retries. You can adjust that to maybe 10 seconds and 2 retries by passing the options `{ retry: 10000, retries: 2 }`.

insideGen, 2017-10-24:
Okay, thanks for the retry options. It works better with that but not all the time. I think you point to the real problem, I use a software to simulate the modbus server, it is likely that the issue comes from it. I will test with a real automaton.

insideGen, 2017-10-26:
I used a `WAGO 750-841` automaton, no error so far. Thanks for your advices!

closed 2017-10-26 by insideGen