node-modbus/stream #17: Detect wrong modbus responses and discard noise bytes

acscode, 2018-03-04:
Hi dresende

Thank you for your great library.

This is my scenario:

- A device slave (programmable automaton) behind NAT communicating through GPRS. Start the communication with the backend and, to keep the connection alive, sends heartbeat periodically to backend (heartbeat is a 0xFF byte).
- A nodejs-backend using modbus-stream in TCP server mode. Once the device-slave establish the TCP connection, the backend send modbus messages (ReadHoldingRegisters, WriteCoil,...) to device-slave.

The issue is: the modbus-stream library doesn't discard the heartbeat from the internal buffer and through exception because of wrong modbus responses.

>  ... `<modbus-response><noise-byte><noise-byte><noise-byte><modbus-response><modbus-response><noise-byte><modbus-response>` ...

Can you add some check code to discard those "heartbeat/noise bytes"?

I've been looking at your code and i think the best place to add this check code is:

lib/transport/tcp.js (line 73) (inside Transport.prototype.unwrap function):

and a simple check code, based on "Protocol Identifier" is always 0x0000 (2th WORD of modbus response), would be:
```
for(; (this.buffer.length >= 6);) {
	if (this.buffer.readUInt16BE(2) !== 0) {
		this.buffer = this.buffer.slice(1, this.buffer.length);
	} else { 
		break;
	}
}
```
What do you think?

Thank you

dresende, 2018-03-07:
I have a similar scenario with hundreds of devices connected to a server and what we do is instead of the 1 byte heartbeat, we pass a simple modbus package with a function code that means heartbeat (out of the defined ones in the standard). This translates to about a minimum of 6 bytes instead of 1. For my scenario it's fine because the heartbeat occurs every 5 minutes or so (I more or less trust the TCP session over GPRS).

If you want to keep that 1 byte heartbeat, I can copy the implementation of the `maxDataInterval` option in the serial transport. What it does is define a maximum time between data. If that time is reached, the current buffer is interpreted and then discarded.

What's the frequency of the heartbeat?

dresende, 2018-03-07:
https://github.com/node-modbus/stream/blob/master/lib/transport/serial.js#L49-L53

acscode, 2018-03-08:
The heartbeat frequency is 1 minute from last sent message (heartbeat, ReadHoldingRegisters,...) We know from experience that Amazon/Azure close idle connections after 4 minutes or less.

Your first solution in which devices send periodically modbus-response like heartbeat is better. I didn't know that I could do that. We've done that and works fine.

Thank you very much

dresende, 2018-03-08:
Glad it worked 😃

closed 2018-03-08 by dresende