node-modbus/stream #18: [Help] How to get incoming-data ?

taimaiduc, 2018-04-11:
var modbus = require("modbus-stream");
```
modbus.tcp.server({ debug: "server" }, (connection) => {
connection.readCoils({from: 1, to: 40}, (err, info) => {
   console.log(info.response.data);
    });
}).listen(502, () => {
    // ready
});
```
**info.response.data is incoming-data ?**

dresende, 2018-04-11:
Yes, it's the response data from your `readCoils` request. If should be an array of 1's and 0's from address 1 to 40.

dresende, 2018-04-13:
I'm not sure I follow. If you want to look at the PDU, the raw data, check `pdu` property.

```js
const modbus = require("modbus-stream");

modbus.tcp.server({ debug: null }, (connection) => {
    connection.readCoils({ from: 3, to: 7 }, (err, info) => {
        console.log("PDU", info.pdu);
        console.log("response", info.response.data);
    });
}).listen(12345, () => {
    modbus.tcp.connect(12345, { debug: null }, (err, connection) => {
        connection.on("read-coils", (request, reply) => {
            console.log("PDU", request.pdu);
            reply(null, [ 1, 0, 1, 0, 1, 1, 0, 1 ]);
        });
    });
});
```

dresende, 2018-04-13:
If you're debugging something, you can change the `debug` property to a text string and see the full packet.

closed 2018-04-17 by taimaiduc