node-modbus/stream #20: slaveId working?

robynjayqueerie, 2018-07-12:
Just trying to get a serial connection on a Raspberry Pi going to a modbus slave with address 10. I know the address is correct and 9600 baud etc because I can connect using another program on the Pi to the device.
I have the following code for modbus-stream
modbus.serial.connect("/dev/ttyUSB0",{slaveId:10},(err,connection)=> etc
The connection appears to be set up as I can look at the connection parameter and it looks OK, the transport variable always contains slaveId:1 regardless of what I put in the setup object. Am I doing something wrong here or does the slaveId setting not work?
Thanks
Robyn

wongni, 2018-07-19:
Try 'extra: { unitId: 10 }' instead of 'slaveId: 10'. For example, 
```
    connection.readDeviceIdentification({ type: "BasicDeviceIdentification", id: "ProductName", extra: { unitId: 255 } }, (err, res) => {
        if (err) throw err;

        console.log(res); // response
    })
```

robynjayqueerie, 2018-07-19:
Thanks will try, is that documented somewhere because I have not seen anything to that effect anywhere

robynjayqueerie, 2018-07-19:
Nah, sorry that does not work, any other suggestions?

wongni, 2018-07-20:
Can you paste your source code here?

robynjayqueerie, 2018-07-20:
This is the connect function as I modified it. The connection still worked with my slave device which had an address of 1, so I assumed that setting the address here to 10 did not work. Thanks

function modbusMaster(){
    // Provides a master connection to slave devices
    modbus.serial.connect("/dev/ttyUSB0",{debug:"automation-123",extra:{unitId:10}},(err,returnConnection)=>{
        if(err)console.log('An error occurred in setting up the mobus master connection');
        masterConnection = returnConnection;

        masterConnection.on('error',(err)=>{
            // evidently transport errors can happen
            console.log('A Serial Modbus error occurred :',err);
        })

        masterConnection.on('close',(err)=>{
            // evidently close errors can happen - do I need to inform the boss?
            console.log('Serial Modbus close error occurred :',err);
        })
    });
}

dresende, 2018-07-23:
Are you sure you're using the latest version? Take a look at this change almost a year ago: https://github.com/node-modbus/stream/commit/13943c7362a4857c48ac6f007de9d02ffebe728f

dresende, 2018-07-23:
You should just use:

```js
function modbusMaster(){
    // Provides a master connection to slave devices
    modbus.serial.connect("/dev/ttyUSB0",{debug:"automation-123",slaveId:10},(err,returnConnection)=>{
        if(err)console.log('An error occurred in setting up the mobus master connection');
        masterConnection = returnConnection;

        masterConnection.on('error',(err)=>{
            // evidently transport errors can happen
            console.log('A Serial Modbus error occurred :',err);
        })

        masterConnection.on('close',(err)=>{
            // evidently close errors can happen - do I need to inform the boss?
            console.log('Serial Modbus close error occurred :',err);
        })
    });
}
```

You have debug lines like this, just see what they have or post them here, I'll convert them.

robynjayqueerie, 2018-07-23:
Thanks @dresende , I was using the alternative version in that example because it was suggested by @wongni However the original version of the function modbusMaster() i created is identical to the one you have listed here. After finding that the change suggested by @wongni did not work I returned the code to the version I started with, which is what you say should work. But it doesn't.
As far as latest modbus-stream version is concerned, I just installed it using the npm command as specified. When I go now and look at the package.json associated with it I see the first lines
"_from": "modbus-stream",
"-id": "modbus-stream@0.40.1"  and again in the last line
"version": "0.40.1", this seems to be the latest version as far as I can determine.
I am running the code on a raspberry pi using stretch. It works fine if I make my slave Id = 1 so I don't think it has anything to do with the pi.
I can halt the code at the masterConnection= line and look at the masterConnection object, within the transport object it always lists slaveId as 1, event though I have used 10 in the options.
I did not work out how to use the debug information there, I did not bother as I could look at the object directly in my development system. Give me a hint as to where I pick up the debug info and I will post it here.
Thanks for your help, I think I have been careful enough in checking my code is identical to that which you have put here.
Robyn

dresende, 2018-07-24:
With the code above, or with yours (if it also has debug enabled), you should have some debug lines starting with `<<` and `>>` which are the protocol data. Can you post it here?

dresende, 2018-07-27:
You're right! I just created a virtual serial and am looking into it.

dresende, 2018-07-27:
This is really a bug, I don't know how I never found this before 😮

dresende, 2018-07-27:
Fixed in version `0.41.0`.

For anyone trying to test and debug serial, you can create a pair of devices to connect (one to master and other to slave) using `socat`.

```sh
socat -d -d pty,raw,echo=0 pty,raw,echo=0
```

closed 2018-07-27 by dresende

robynjayqueerie, 2018-07-28:
@dresende Thank you, I have not tried it yet but will get the later version next week and run it. Glad my bug catching talent was useful :)

commit 7823faa3 (2018-07-27): fixes #20