node-modbus/stream #26: Module not found: Error: Can't resolve 'serialport/package'

hxl-dy, 2018-09-19:
webpack@^3

dresende, 2018-09-19:
What version are you using and in what system are you trying?

hxl-dy, 2018-09-19:
windows 10
webpack@^3

dresende, 2018-09-19:
I'm not sure I can help as I don't use Windows. Are you able to just create a folder and install `serialport` package there?

hxl-dy, 2018-09-19:
I have installed serialport@7.0.2, but the error still occur.

dresende, 2018-09-19:
Never tried above version `6.0.0` and never done it on Windows, can't help you 😢

hxl-dy, 2018-09-19:
fixed: add this config to webpack config:
```javascript
module.exports = {
  resolve: {
    extensions: ['.json', '.ts', '.js'],
  },
}
```
Because [this line](https://github.com/node-modbus/stream/blob/master/lib/driver/serial.js#L1) does not use `.json` explicitly.

closed 2018-09-19 by hxl-dy