node-modbus/stream #27: Hello

Mohamedelhrichi, 2018-10-17:
i have a modbus client with ( modbus unit id = 245 for example) how can i add this in node-modbus stream methods to read values ( coils and inputs )

dresende, 2018-10-17:
If you're using a serial device..

```js
const modbus = require("modbus");

modbus.serial.connect("/dev/ttyUSB0", { slaveId: 245 },(err, connection) => {
    // ...
    // connection.readCoils(...)
});
```

If you're using TCP...

```js
const modbus = require("modbus");

modbus.tcp.connect("192.168.253.78", { slaveId: 245 },(err, connection) => {
    // ...
    // connection.readCoils(...)
});
```

Mohamedelhrichi, 2018-10-18:
tank you a lot for you answer.
I tested this  but i have the same probleem when a add a salveid option  

var modbus = require("modbus-stream");

 function readclient(ip) {
     modbus.tcp.connect(502, ip,{ slaveId: 245} , { debug: "automaton-2454" }, (err, connection) => {
console.log("connected")
connection.readCoils({ address: 0, quantity: 10 }, (err, res) => {
    if (err) throw err;

    console.log(res.response.data); // response
})
});
 }
 readclient("127.0.0.1") ;

 i have receved  this message

Error: GatewayTargetDeviceFailedToRespond
    at Object.exports.error (C:\Users\RAYAN\Desktop\mod\node_modules\modbus-pdu\lib\Exception.js:24:13)
    at Timeout._onTimeout (C:\Users\RAYAN\Desktop\mod\node_modules\modbus-stream\lib\transport\transport.js:93:33)
    at ontimeout (timers.js:482:11)
    at tryOnTimeout (timers.js:317:5)
    at Timer.listOnTimeout (timers.js:277:5)

dresende, 2018-10-18:
`slaveId` does not exist in TCP, only `unitId`.

Mohamedelhrichi, 2018-10-18:
thank you for your answers i rectified my code but it's give the same error 

var modbus = require("modbus-stream");
 function readclient(ip) {
     modbus.tcp.connect( 502, ip,{unitId : 245} ,
         { debug: "automaton-2454" }, (err, connection) => {
console.log("connected")
connection.readInputRegisters({ address: 0, quantity: 10 }, (err, res) => {
    if (err) throw err;

    console.log(res.response.data); // response
})
});
 }

 readclient("127.0.0.1") ;

Error: GatewayTargetDeviceFailedToRespond
    at Object.exports.error (C:\Users\RAYAN\Desktop\mod\node_modules\modbus-pdu\lib\Exception.js:24:13)
    at Timeout._onTimeout (C:\Users\RAYAN\Desktop\mod\node_modules\modbus-stream\lib\transport\transport.js:93:33)
    at ontimeout (timers.js:482:11)
    at tryOnTimeout (timers.js:317:5)
    at Timer.listOnTimeout (timers.js:277:5)

dresende, 2018-10-18:
Do you have a modbus client on your computer? Are you sure it's on port 502 and responds to that request?

Mohamedelhrichi, 2018-10-20:
hello it's OK now it's worked with this code 

`//var modbus = require("modbus-stream");
const modbus = require("modbus-stream");
var fs = require('fs') ;

 var mydata ;
 var mydata2 ;
 var mydata3 ="bonjour" ; 

 function readclient(ip) {
   

    modbus.tcp.connect(ip, { unitId: 245 },(err, connection) => {
        // ...
        connection.readInputRegisters({ address: 259, quantity: 20 }, (err, res) => {
            if (err) throw err;
        
            console.log(res); 
            mydata = res.response.data ;
            // response
            mydata2=res.pdu ;
            
         //  mydata3 = mydata2.toString('utf8') ;
          //  var ret = mydata3.replace(/Buffer-/g,'');
         //   var ret2 = ret.replace(/<-/g,'');
          //  var ret3 = ret2.replace(/>-/g,'');
          //  mydata3 = ret3 ;
          //  console.log(mydata3) ;
            
      
        })
        });
 }

 //,{ slaveId: 1}
 readclient("10.80.15.62") ;
`

Mohamedelhrichi, 2018-10-20:
i want to know how a can use buffer data  and convert it to int array or decimal array

dresende, 2018-10-22:
Modbus does not define any types for data, you need to know how and what they have encoded. Show me the response (`res` or just `res.response.data`) and tell me what it should have there.

Mohamedelhrichi, 2018-10-22:
ok thanks its OK a wondefull module thanks a lot

closed 2018-11-02 by dresende

sarat12, 2019-12-03:
Thanks for the decent peace of software! Is there any chance to change slaveId using the same connection? (I do polling on serial port and want to keep connection opened).

sarat12, 2019-12-03:
Sorry for bothering, just figured that out:
`connection.transport.slaveId = slaveId;`

dresende, 2019-12-06:
😉