node-modbus/stream #34: Correct Parameters to reply to writeMultipleRegisters

EricPublic12, 2019-02-04:
I am trying to create a server and I need to send the reply after the client calls writeMultipleRegisters. 
What are the correct parameters to send back and is there a helper function that will create the response?

dresende, 2019-02-05:
You just need to reply the address of the first register and the number of registers written.

```js
const modbus = require("modbus-stream");

modbus.tcp.server({ debug: null }, (connection) => {
	let one_hundred  = Buffer.alloc(2);
	let one_thousand = Buffer.alloc(2);
	let ten_thousand = Buffer.alloc(2);

	one_hundred.writeUInt16BE(100, 0);
	one_thousand.writeUInt16BE(1000, 0);
	ten_thousand.writeUInt16BE(10000, 0);

	connection.writeMultipleRegisters({ address: 123, values: [ one_hundred, one_thousand, ten_thousand ] }, (err, info) => {
		console.log("response", info.response);
	});
}).listen(12345, () => {
	modbus.tcp.connect(12345, { debug: null }, (err, connection) => {
		connection.on("write-multiple-registers", (info, reply) => {
			console.log("request", info.request);

			console.log("one hundred", info.request.values[0].readUInt16BE(0));
			console.log("one thousand", info.request.values[1].readUInt16BE(0));
			console.log("ten thousand", info.request.values[2].readUInt16BE(0));

			return reply(null, info.request.address, info.request.quantity);
		});
	});
});
```

closed 2019-02-05 by dresende

dresende, 2019-02-06:
The responses are always based on standard. For `writeSingleRegister` you should return the address and the value written (instead of the quantity). Basically you just reply the same thing you receive.