node-modbus/stream #35: Why does the example show the server sending a request to the client?

jacobq, 2019-02-06:
In Modbus(TCP), the terms "server" & "slave" refer to devices that do not initiate contact with other devices but rather only respond to request they receive. The terms "client" & "master" refer to devices that initiate requests. So why does [the first example](https://github.com/node-modbus/stream#example) have the server (slave) send a request (`read-coils` address 3-7) to a client (master)? This seems backwards. 

> This is my current test.js file. It creates a client and a server network socket and the server requests coils as soon as the client connects.

dresende, 2019-02-06:
We could've used other names but we wanted to avoid master and slave and just use the terms used in networking for a server (which listens for a connection) and a client (which connects to). Also, you can initiate connections from any endpoint, there's no restriction on that, to enable developers to use modbus in more scenarios.

jacobq, 2019-02-06:
> Also, you can initiate connections from any endpoint, there's no restriction on that, to enable developers to use modbus in more scenarios.

Could you cite a corresponding standard regarding this? I understand that TCP allows this and perhaps your library does too, but I do not believe it is part of the Modbus standards. For example, [MODBUS Application Protocol Specification V1.1b3](http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) says on page 4 (emphasis added):
> The MODBUS application data unit is built by the **client** that **initiates** a MODBUS transaction.  ... The MODBUS application protocol establishes the format of a **request** initiated by a **client**.

Furthermore, all of the sequence diagrams show the flow of data being initiated by the client.  

The [MODBUS MESSAGING ON TCP/IP IMPLEMENTATION GUIDE V1.0b](http://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf) also talks about this in the section titled "1.2 Client / Server Model" (page 2, emphasis added):

> This client / server model is based on four type of messages: ...
> A MODBUS **Request** is the message sent on the network by the **Client** to **initiate** a transaction,
> A MODBUS Indication is the Request message received on the Server side,
> A MODBUS **Response** is the Response message sent by the **Server**,
> A MODBUS Confirmation is the Response Message received on the Client side

dresende, 2019-02-08:
> unless you mean that each end is acting as both client and server

That's what I wanted to state. This library follows protocol specification (message format), but does not enforce how you initiate connections or which endpoint is responsible for making requests.

I use this library for example on a scenario where devices connect to a centralised server but it's the server that then uses the connection to make requests.

dresende, 2019-02-08:
Where you see `modbus.tcp.server` in the example, we should've used `modbus.tcp.listen`, following the network API of node.js (and others), to indicate that we're just binding to a local port.