node-modbus/stream #37: How can one specify a `unit ID` to associate with a request?

jacobq, 2019-02-14:
It seems that this implementation assumes a "Unit identifer" (see [3.1.2 MODBUS On TCP/IP Application Data Unit ](http://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf#page=5)) value of `1` by default. How can this by changed by the caller / consuming application? Is there some property of the `extra` options that can do this?

jacobq, 2019-02-14:
I took a guess of using `readCoils({ ... extra: { unitId: 123 }})` (based on [similar syntax in `quirks.js`](https://github.com/node-modbus/stream/blob/dea86d7ea58f1c39564e8db35cbbeaa72feec172/test/integration/quirks.js#L24)) and it worked! Would you please consider adding this to the documentation?

(If you're not interested in clarifying that, feel free to close the issue since I've at least figured out what I needed.)

dresende, 2019-02-14:
You're right, there's not much information. I'll add that to one of the examples of the README until I find a better way of listing options.

closed 2019-02-14 by dresende