node-modbus/stream #38: Syntax for generic function

jacobq, 2019-02-28:
Similar to https://github.com/node-modbus/stream/issues/34 I'm confused about the different signatures that need to be used for `reply`. It seems to depend on the function code.

For example, in my application I define a "modbus map" to associate handler functions with ROB & ROR read operations (function code 2 & 4) as well as RWB & RWR read/write operations (codes 1, 3, 5, 6, 15, and 16).

```js
modbus.tcp.server(serverOptions, (connection: any) => {
    // ...
    connection.transport.on('request', (eventName, transaction, reply) => {
        const request = transaction.request;
        const size = request.quantity;
        const firstAddr = request.address;
        const lastAddr = firstAddr + size - 1;
        // (call appropriate handlers)
        // ...
        // This is the part that seems weird / opaque to me.
        // Why isn't there a function that will just accept `result` and format it appropriately
        // (since the transaction contains the information needed)?
        if (eventName === 'write-single-register') {
            reply(null, firstAddr, result[0]);
        }
        else if (eventName === 'write-multiple-registers') {
            reply(null, firstAddr, result);
        }
        else {
            reply(null, result);
        }
    })
    // ...
}
```

dresende, 2019-03-01:
Yes, the `reply` assumes a first (fixed) argument for an error, and then the remaining arguments depend on the function code and what the standard defines for a response.

jacobq, 2019-03-01:
> Yes, the `reply` assumes a first (fixed) argument for an error, and then the remaining arguments depend on the function code and what the standard defines for a response.

OK, thank you for clarifying that. Is it the same regardless of whether `reply` is obtained from a specific event like the `read-coils` event vs. from the lower level `request` event?

```js
connection.on("request", (request, reply) => {
    // ...
});
connection.on("read-coils", (request, reply) => {
    // ...
    // does this `reply` have the same signature/contract as the one above?
});

```

dresende, 2019-03-01:
> Is it the same regardless of whether `reply` is obtained from a specific event like `read-coils` vs. from the lower level `request` event?

I believe yes. But perhaps, in that case, it makes more sense to be able to send a raw response. You could just ignore the `reply` and write a response yourself, but it would be easier.

jacobq, 2019-03-01:
OK, thanks for the reply. I will plan to just keep using the `request` event and it's `reply` function as I have it (like above -- seems to be working OK). It just feels a little non-intuitive because the library code does some of the protocol work (e.g. ID & function code / exception code) but not all of it (e.g. start address & number of coils/registers), so I have to dig into the library code to see what I need to do. For example, should I be sending "one big buffer" or separate arguments for, say, start address and array of data. 

> You could just ignore the `reply` and write a response yourself

Would that look something like this: [`transport.write(<Buffer>)`](https://nodejs.org/api/stream.html#stream_writable_streams)?

dresende, 2019-03-02:
Yes.

jacobq, 2021-06-07:
Closing due to inactivity

closed 2021-06-07 by jacobq