node-modbus/stream #40: I don't get a response, but debug information

uNki23, 2019-03-07:
Hi!

I'm able to establish a connection to a Siemens S7 via Modbus TCP. 

I'm trying to read the holding registers. If I run the application I get the "Connected." info and the debug information with the holding register data. But I don't get the res object of the function call. So basically I can't do anything with the data since it just displays the debug information.

`const modbus = require("modbus-stream");

modbus.tcp.connect(502, "192.168.0.1", { debug: "server" }, (err, connection) => {
    if (err)
        console.error('Connection error: ', err)

    console.log('Connected.\n')

    connection.readHoldingRegisters({ address: 0, quantity: 20, extra: { unitId: 1 } }, (err, res) => {
        if (err)
            console.error('Error while reading holding registers: ', err)
        console.log('Response: ', res); // <-- THIS IS NEVER DISPLAYED
    })

    connection.on('error', err => {
        if (err.message != 'write EPIPE')
            console.log('Unknown error:', err.message)
    })
});`

dresende, 2019-03-07:
That's very weird, we communicate with S7 all the time. Have you tried debugging using another software to see what/how it responds?

uNki23, 2019-03-08:
Sorry, error was on our end. Connection to the S7 was corrupted. This thing can be closed / deleted.

Awesome library! 👍

closed 2019-03-08 by uNki23