node-modbus/stream #41: Supporting SSL / TLS

ChristianF1, 2019-03-21:
I am not able to find any information regarding the support of SSL / TLS. I am not able to see any specific security options on the tcp transport / connect options

Is it as simple as specifying the SSL port number instead?

dresende, 2019-03-21:
> Is it as simple as specifying the SSL port number instead?

Yes an no. It's not just specifying the SSL port, but with a little effort you may solve this for you. We never added SSL as we don't found any use case and real world example (sadly) to test it out.

Bear in mind this is from the top of my head, never tested.

```js
const modbus = require("modbus-stream");
const tls    = require("tls");

let connection = tls.connect({ tls_options }, () => {
	let stream = new (modbus.stream)(modbus.transports.tcp.prepare(connection));

	// do something with stream
});
```

If you find this useful and working, let me know. I might add it to the module.

ChristianF1, 2019-03-21:
Tx for the suggestion. We dont have an immediate need, but want to make sure we have the option in the future. I will see if we can do a test.

closed 2019-03-21 by ChristianF1