node-modbus/stream #43: Node 12 compatibility

zamu87, 2019-05-17:
The version of serialport included in package.json (<7) is not compatible with node 12 and generates a lot of errors during installation. Have you already planned the update of that package?

dresende, 2019-05-17:
No, I use this module quite a lot on arm devices so I need a simpler serialport module. I'm not sure why I added the `<7` but perhaps it was because of the errors.

Does version `>=7` work with the module?

zamu87, 2019-05-17:
I've not tested it with serialport 7.x because I haven't any serial modbus device but only tcp.

dresende, 2019-05-17:
Then you should be fine without that dependency at all.

sarat12, 2019-12-04:
> No, I use this module quite a lot on arm devices so I need a simpler serialport module. I'm not sure why I added the `<7` but perhaps it was because of the errors.
> 
> Does version `>=7` work with the module?

I've just tested it with 8.0.5. It mostly works. writeSingleCoil() and writeSingleRegister() look ok, readHoldingRegisters() and readInputRegisters() too. But readCoils() and readDiscreteInputs() are not. They always return array of length Math.ceil(quantity/8)*8 and trails with zeros, so if { address: 1, quantity: 5 } even if all 8 coils are 1s it returns [1,1,1,1,1,0,0,0].

sarat12, 2019-12-08:
> I've just tested it with 8.0.5. It mostly works. writeSingleCoil() and writeSingleRegister() look ok, readHoldingRegisters() and readInputRegisters() too. But readCoils() and readDiscreteInputs() [...]

hmm.. I've switched back to 7 and i see the same. Is that expected?