node-modbus/stream #50: Error: GatewayTargetDeviceFailedToRespond

sexbee, 2019-11-13:
receive data from TCP client : 

{ Error: GatewayTargetDeviceFailedToRespond
    at Object.exports.error (/Users/mac/Desktop/RLLP-PCL5new/node_modules/modbus-pdu/lib/Exception.js:24:13)
    at Timeout._onTimeout (/Users/mac/Desktop/RLLP-PCL5new/node_modules/modbus-stream/lib/transport/transport.js:92:33)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) code: 11 }

I cant find cause?

sexbee, 2019-11-13:
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at TCP.onread (net.js:622:25)

viatick, 2019-11-17:
[image]

error : Error: GatewayTargetDeviceFailedToRespond
result : undefined

when reading at address 0?