node-modbus/stream #53: Modbus : get Discover all the devices and generate the data model

yboujraf, 2019-12-23:
Dear @jhillacre , @dresende , @jacobq , @buffcode , @insideGen 

I am working on a proprietary protocol translator from /to modbus.

I need to discover all the modbus device master/slave and get the list of parameters they are able to support (finally the data model) for each devic discovered.

It's our data model for one parameter it could be string, boolean, real, integer, ....

```
{

                    "number": 0,
                    "identifier": "ParameTypeInteger",
                    "description": "",
                    "value": 0,
                    "minimum": -4096,
                    "maximum": 0,
                    "access": "readWrite",
                    "format": "%8.2f°",
                    "enumeration": "",
                    "factor": 32,
                    "isOnline": true,
                    "formula": "1*\n1/",
                    "default": -1024,
                    "type": "integer"
                }               {

                    "number": 1,
                    "identifier": "ParameTypeReal",
                    "description": "",
                    "value": 0,
                    "minimum": 0,
                    "maximum": 4096,
                    "access": "readWrite",
                    "format": "%8.2f°",
                    "enumeration": "",
                    "factor": 1,
                    "isOnline": true,
                    "formula": "10*\n10/",
                    "default": 100,
                    "type": "real"
                },
                {
                    "number": 2,
                    "identifier": "ParameTypeString",
                    "description": "",
                    "value": "text",
                    "access": "readWrite",
                    "enumeration": "Yes\nNo\nOk",
                    "isOnline": true,
                    "default": "default",
                    "type": "string"
                },
                {
                    "number": 3,
                    "identifier": "ParameTypeEnumeration",
                    "description": "",
                    "value": 0,
                    "access": "readWrite",
                    "enumeration": "Line\nMic+48V\nMic",
                    "isOnline": true,
                    "default": 0,
                    "type": "enum"
                },
                {
                    "number": 4,
                    "identifier": "ParameTypeBoolean",
                    "description": "",
                    "value": true,
                    "access": "readWrite",
                    "isOnline": true,
                    "default": false,
                    "type": "boolean"
                }
```

How we could do with you code to generate a json file with all the devices ?

Best regards
Youssef

jacobq, 2019-12-23:
It sounds like you are asking for someone else to write your code for you. Do you have a specific question about this library? It basically just creates the Modbus messages. You have to write the logic for what to send, how to respond, etc.

closed 2020-01-03 by dresende

yboujraf, 2020-01-03:
Dear @dresende 

I have never request to somebody to write anything. Apology if you have understood like this.

I have never used Modbus and I didn't know that modbus could discover and get a json file with data model.

I will try to find a device with modbus and will inform you.

Best Wishes and best regards
Youssef

jacobq, 2020-01-03:
> I have never used Modbus...

I recommend reviewing the specification to get familiar with the protocol before designing an API that uses it.

https://en.wikipedia.org/wiki/Modbus
http://www.modbus.org/specs.php
http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf

Basically Modbus provides a way to transmit data onto a communications bus, control access to the medium (e.g. defines timing for serial lines), identify where you'd like to send that data (either address a specific device or broadcast), and verify data integrity (CRC). The messages are either "read" or "write" and associated with either "single" or "multiple" "coils" (bits) or "registers" (16-bit words).

yboujraf, 2020-01-03:
Dear @jacobq 

Thanks you very much, Will add it to my new sprint to learn more in details the modbus.

Best Regards,
Yuussef