node-modbus/stream #55: compatibility Node version 12.14.0

simsimo, 2019-12-29:
it's not compatible with the lastest version of nodejs 12.14.0 (npm V6.13.4 )

dresende, 2020-01-03:
Do you have a log where you found incompatibility?

mschloapps, 2020-01-06:
I have this issue trying to install:

3746 verbose stack Error: serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
3746 verbose stack Exit status 1
3746 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:332:16)
3746 verbose stack     at EventEmitter.emit (events.js:210:5)
3746 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
3746 verbose stack     at ChildProcess.emit (events.js:210:5)
3746 verbose stack     at maybeClose (internal/child_process.js:1021:16)
3746 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
3747 verbose pkgid serialport@6.2.2
3748 verbose cwd C:\Users\schlo\Projects\Node.js
3749 verbose Windows_NT 10.0.18363
3750 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "i" "modbus-stream"
3751 verbose node v12.14.0
3752 verbose npm  v6.13.4
3753 error code ELIFECYCLE
3754 error errno 1
3755 error serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
3755 error Exit status 1
3756 error Failed at the serialport@6.2.2 install script.
3756 error This is probably not a problem with npm. There is likely additional logging output above.
3757 verbose exit [ 1, true ]

bcatalin, 2020-01-14:
I confirm this for the same version of the nodeJs. serialport@6.2.2 is the problem. Installing the serialport@8.0.4 and manual install of the package is an workaround I've found. I am also using the TCP and I can't confirm that the serial connection is working.

paratim, 2020-05-12:
> I confirm this for the same version of the nodeJs. serialport@6.2.2 is the problem
Confirmed that this is still a problem for node v12.16

leocotes, 2020-07-16:
I've the same problem. 
I already install  serialport@8.0.4, but until error to install modbus-stream.

NODE 12.18.0
NPM 6.13.7

jnavila, 2020-09-14:
Same here. @bcatalin . Could you explain how you managed to install this package manually, please?

jacobq, 2020-09-21:
> Same here. @bcatalin . Could you explain how you managed to install this package manually, please?

You can do it with `yarn`'s [selective dependency resolutions](https://classic.yarnpkg.com/en/docs/selective-version-resolutions/) like this (in `package.json`):

```json
{
  // ... dependencies and all that ...
  "resolutions": {
    "**/serialport": "^9.0.1"
  }
}
```