node-modbus/stream #58: QUESTION How to get device data once a second?

Serhioromano, 2020-03-03:
The code I can understand it is simple. But my problem is connection. Fo instance I have something like 

```js
setInterval(() => {
    // get device infor nere
}, 2000);
```

1, In the section where I collect data from the device, do I call connect every time or I should connect before the interval and then only read registers? Something like this

```js
let run = async () => {
    let conn = await modbus.tcp.connect(502, "134.2.56.231", { debug: "automaton-2454" });
    setInterval(() => {
        conn.readCoil();
    }, 2000);
}
```

Or like this

```js
setInterval(() => {
    modbus.tcp.connect(502, "134.2.56.231", { debug: "automaton-2454" }, (err, connection) => {
        connection.readCoil();
    });
}, 2000);
```

2. If I can do the first method then how do I catch en error if the connection gets lost?

dresende, 2020-03-04:
Assuming your `setInterval` is really 2 seconds, I believe it's better to go with option 1 (keep connection open). You can reopen by listening for the `close` event on the object (Modbus Stream) returned by the connect method.

Serhioromano, 2020-03-04:
An example would be very handy. Very hard to guess how it should be organized. My question is still there. Can you sketch a tiny example?

dresende, 2020-03-04:
Something like this perhaps. Haven't tested.

```js
let conn     = null;
let get_conn = async () => {
    if (conn) return conn;

    conn = await modbus.tcp.connect(502, "134.2.56.231", { debug: "automaton-2454" });

    conn.on("close", () => {
        // will reconnect when requested
        conn = null;
        // instead, you could just do " conn = get_conn(); " to reconnect immediately
    });

    return conn;
};

setInterval(() => {
    let instance = await get_conn();

    instance.readCoil();
}, 2000);
```

Serhioromano, 2020-03-04:
Looks fine. Enough to grasp an idea. Where do I put flag to keep a connection open? And what ` { debug: "automaton-2454" }` means? Should I set it or not?

dresende, 2020-03-05:
> Where do I put flag to keep a connection open?

It should automatically keep it open.

> And what { debug: "automaton-2454" } means?

That is for debug lines (not sent, just console). Disable by passing `{ debug: null }`.

closed 2020-03-13 by dresende