node-modbus/stream #68: Won't install serialport@6.2.2 dependency on Mac OS 11.3.1

ricozinn, 2021-05-13:
Hi, just before installing the last Mac OS update (just last week) this worked fine, but after installing it I can't get it to work and it gets stuck on the serialport dependency.  Note that I can install the latest serialport releast v9.x just fine.  It's the serialport@6.2.2 you require that breaks it.

Any ideas?  Thanks in advance for any help!
```
prebuild-install WARN install No prebuilt binaries found (target=16.1.0 runtime=node arch=x64 platform=darwin)
(node:19382) [DEP0150] DeprecationWarning: Setting process.config is deprecated. In the future the property will be read-only.
(Use `node --trace-deprecation ...` to show where the warning was created)
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp:41:48: error: too few arguments to function call, single argument 'context' was not specified
  v8::String::Utf8Value path(info[0]->ToString());
                             ~~~~~~~~~~~~~~~~~ ^
/Users/richard.zinn/Library/Caches/node-gyp/16.1.0/include/node/v8.h:3019:44: note: 'ToString' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
                                           ^
../src/serialport.cpp:48:53: error: too few arguments to function call, single argument 'context' was not specified
  v8::Local<v8::Object> options = info[1]->ToObject();
                                  ~~~~~~~~~~~~~~~~~ ^
/Users/richard.zinn/Library/Caches/node-gyp/16.1.0/include/node/v8.h:3031:44: note: 'ToObject' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                           ^
../src/serialport.cpp:95:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(2, argv);
                 ^
../../../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/serialport.cpp:113:53: error: too few arguments to function call, single argument 'context' was not specified
  v8::Local<v8::Object> options = info[1]->ToObject();
                                  ~~~~~~~~~~~~~~~~~ ^
/Users/richard.zinn/Library/Caches/node-gyp/16.1.0/include/node/v8.h:3031:44: note: 'ToObject' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                           ^
../src/serialport.cpp:150:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(1, argv);
                 ^
../../../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/serialport.cpp:188:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(1, argv);
                 ^
../../../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/serialport.cpp:231:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(1, argv);
                 ^
../../../../nan/nan.h:1741:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../../../nan/nan.h:106:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/serialport.cpp:250:53: error: too few arguments to function call, single argument 'context' was not specified
  v8::Local<v8::Object> options = info[1]->ToObject();
                                  ~~~~~~~~~~~~~~~~~ ^
/Users/richard.zinn/Library/Caches/node-gyp/16.1.0/include/node/v8.h:3031:44: note: 'ToObject' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
                                           ^
../src/serialport.cpp:285:18: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  data->callback.Call(1, argv);
[... 76 lines cut ...]
../src/serialport.cpp:460:17: error: no member named 'Handle' in namespace 'v8'
  void init(v8::Handle<v8::Object> target) {
            ~~~~^
../src/serialport.cpp:460:36: error: use of undeclared identifier 'target'
  void init(v8::Handle<v8::Object> target) {
                                   ^
../src/serialport.cpp:460:43: error: expected ';' after top level declarator
  void init(v8::Handle<v8::Object> target) {
                                          ^
                                          ;
8 warnings and 13 errors generated.
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (node:events:365:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
gyp ERR! System Darwin 20.4.0
gyp ERR! command "/usr/local/Cellar/node/16.1.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/richard.zinn/Sites/zense_epv2/gui/node_modules/modbus-stream/node_modules/serialport
gyp ERR! node -v v16.1.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 

```

thirstyfish, 2022-01-25:
Hi @ricozinn , having the same issue on M1 Mac running on Monterey.
Did you manage to resolve it? Thx

thirstyfish, 2022-01-25:
> Hi @ricozinn , having the same issue on M1 Mac running on Monterey. Did you manage to resolve it? Thx

For posterity, this workaround seems to work:
https://github.com/node-modbus/stream/issues/55#issuecomment-696224704