node-modbus/stream #69: Query three different registers

Aaqu, 2021-08-12:
I am new to modbus stream, how to query three different registers (e.g. 1026, 1032, 1552) on one connection. Need to use async / await?

Aaqu, 2021-08-12:
I tried but it doesn't work for me
https://github.com/node-modbus/stream/issues/58#issuecomment-594571454

dresende, 2021-08-12:
Yes, you need something to handle multiple requests. You could use `quantity` parameter (see README) but in your case you would have to require 26 records.

closed 2021-08-12 by dresende

Aaqu, 2021-08-12:
Will an example be added to the documentation in the future?

dresende, 2021-08-12:
It's there, you can just copy the readCoils example and change to holding registers or input registers.

```js
// for a serial connetion
modbus.serial.connect("/dev/ttyS123", { debug: "automaton-123" }, (err, connection) => {
    if (err) throw err;

    connection.readHoldingRegisters({ address: 1026, quantity: 26 }, (err, res) => {
        if (err) throw err;

        console.log(res.response.data); // response
    })
});
```

Aaqu, 2021-08-12:
> It's there, you can just copy the readCoils example and change to holding registers or input registers.
>
> ```js
> // for a serial connetion
> modbus.serial.connect("/dev/ttyS123", { debug: [...]

Sorry i thought async / await, if i need 1020 and 1520 i don't want to download all 500 registers

dresende, 2021-08-12:
Async / await is not supported for now (the code does not use Promises). What you can do instead is use module async with `async.series`.

```js
// for a serial connetion
modbus.serial.connect("/dev/ttyS123", { debug: "automaton-123" }, (err, connection) => {
    if (err) throw err;

    async.series([
        (next) => {
            connection.readHoldingRegisters({ address: 1020 }, (err, res) => {
                if (err) return next(err);

                console.log(res.response.data); // response

                return next();
            });
        },
        (next) => {
            connection.readHoldingRegisters({ address: 1520 }, (err, res) => {
                if (err) return next(err);

                console.log(res.response.data); // response

                return next();
            });
        },
    ], (err) => {
        // move on..
    });
});
```

Aaqu, 2021-08-12:
Thank you for your help now I understand

Aaqu, 2021-08-13:
What do you think about this solution?
`
function newTcpDataStream(port, ip, id, register, querys) {
    return new Promise(function (resolve) {
        setTimeout(function () {
            modbus.tcp.connect(port, ip, { debug: "none", unitId: id }, (err, connection) => {
                connection.readInputRegisters({ address: register, quantity: querys }, (err, data) => {
                    connection.close();
                    resolve(data.response.data);
                });
            });
        }, 10);
    });
}

async function streamTcp() {

    const data1 = await newTcpDataStream(500, '192.168.1.1', 1, 1024, 5)
    const data2 = await newTcpDataStream(500, '192.168.1.1', 1, 1536, 2)
    const data3 = await newTcpDataStream(500, '192.168.1.1', 1, 1557, 2)

    console.log(data1)
    console.log(data2)
    console.log(data3)

}

streamTcp();
`

Aaqu, 2021-08-13:
Wow! without setTimeout also works again thanks for help

dresende, 2021-08-13:
That's good. You can also check for `err` and call `reject(err)` and you have full support for async (in this case, wrap your `streamTcp` code inside a try/catch).

Aaqu, 2021-08-16:
Does `connectTimeout` also work for `readInputRegisters`?

dresende, 2021-08-17:
Try option `retries` (number of times to repeat request before quitting) and `retry` (timeout before trying again).

```js
modbus.tcp.connect(port, "host", {
    retry   : 1000, // milliseconds = 1 second
    retries : 4,    // 1 try + 4 retries = 5 attempts
});
```