node-modbus/stream #71: Convert response from ReadHoldingRegisters to Decimal

eskstyle, 2022-03-09:
I'm getting this response <Buffer 00 28>, how can i convert to dec?

Thanks

var modbus = require("modbus-stream");

modbus.tcp.connect(502, "10.10.10.8", (err, connection) => {
  if (err) {
    console.log("Falha!");
    throw err;
  }
  console.log("Conectado");

  connection.readHoldingRegisters({ address: 1, quantity: 1 }, (err, res) => {
    if (err) throw err;

    console.log(res.response.data[0]); // response

    console.log(new Uint8Array(res.response.data[0]));
  });
});

dresende, 2022-03-09:
Modbus does not enforce types, so you need to know the type of data. Usually, for unsigned integers of 2 bytes, modbus implementers use uint16be. You can do this:

```js
console.log(res.response.data[0].readUInt16BE(0));
```

eskstyle, 2022-03-10:
> Modbus does not enforce types, so you need to know the type of data. Usually, for unsigned integers of 2 bytes, modbus implementers use uint16be. You can do this:
> 
> ```js
> console.log(res.response.data[0].readUInt16BE(0));
> ```

It worked perfectly, thank you very much!!

closed 2022-03-10 by alancsdev