node-modbus/stream #74: How to catch tcp transport errors

boatbodger, 2022-12-06:
I am writing a client which needs to recover from temporary inability to connect to the modbus server the bare bones of what I'm trying to do is as follows 

```
// Variables into which the data read from the MODBUS is put, available for forwarding to any web consumer
var meterVolts
const U16_METER_V = 33128		//Register Volts at meter in tenths of a Volt
//****************************************************************
//
//       MODBUS Handler etc (functions followed by some inline code)
//
//****************************************************************

const modbus = require("modbus-stream");
const util = require('util')

function getValuesPromise(connection,req){
	return new Promise ((resolve, reject) => {
		if (connection == null)
			reject('Connection found to be closed')
		connection.readInputRegisters(req,(err,res) => {
			if (err)
				reject(err)
			else 
				resolve(res.response.data)
		})
	})
}

async function getInverterInfo(connection){

	connection.transport.on("error", function (err){
			console.log("transport error",err)
			//raise reconnect request event
			return
	})
	connection.transport.on("close", function(err){
		console.log("transport close",err )
		return
	})

	//Meter voltage as a way to test manipulation of a U16 result
	meterVolts =  await getValuesPromise(connection,{address:U16_METER_V, quantity: 1, extra: { unitId: 1}}).catch(err => {console.log(err)})

	setTimeout(() => {getInverterInfo(connection)},10000)	//Run again after ten seconds
}

	modbus.tcp.connect(8899,"192.168.100.242",{debug: null}, (err, connection)=> {
		if (err != null){
			console.log('could not connect')
			console.log(err)
		};

		getInverterInfo(connection)	//This kicks a repeat timer.

	});
`
```

However, I'm finding that if I reset the server during the 'wait' time, instead of cleanly catching either a "close" or an "error" event, I see the following in my console:
``
Error: GatewayTargetDeviceFailedToRespond
    at exports.error (/home/solar/node_modules/modbus-pdu/lib/Exception.js:24:13)
    at Timeout._onTimeout (/home/solar/node_modules/modbus-stream/lib/transport/transport.js:93:33)
    at listOnTimeout (node:internal/timers:564:17)
    at process.processTimers (node:internal/timers:507:7) {
  code: 11
}
``
Can anyone help me understand what I'm doing wrong?

dresende, 2022-12-06:
That error is an exception when transport has given up trying to send the request and not having an answer. Two things:

1. You don't need to listen for `connection.transport` events, those are proxied to `connection` so you can listen directly on them;
2. Your `reject('Connection found to be closed')` shouldn't have a return? This does not solve the issue, it's just something that catch my eye.

I'm not sure what's going on, I would have to simulate your scenario.

boatbodger, 2022-12-06:
Thank you very much for your response.
re 1. yes, I'm aware that transport has given up, but surely that error should flow through the proxy you mentioned to my code - not simply trash the script?  That's what I'm not understanding.  I did initially try to catch the error at the connection level, but that produced syntax errors - I'll re-try, though, I may have got something else wrong.

re 2. Yes, that return should not be there.  I was clutching at straws

vaniatoperich, 2023-03-21:
Any success on this? I am also facing the same problem with this exception and I do not get any event on connection (I am listening for close and error as documented)

vaniatoperich, 2023-03-22:
What I cam up with on the end is to emit the event my self when I catch the GatewayTargetDeviceFailedToRespond error. That seems to be working