node-modbus/stream #75: Modbus ERROR: GatewayPathUnavailable

XavegX367, 2022-12-12:
Hello, I am having an issue. I am retrieving a value from modbus, with an interval set to a few seconds, but after a few times it randomly returns the error: write EPIPE and after that it returns read: ECONNRESET or: GatewayPathUnavailable. Is there a way to reset the Node server once this happens, because after a restart the function inmediately runs fine again. I don't really know what I am doing wrong, since I am new to Node servers. Here is the code:

`
const modbus = require('modbus-stream');

module.exports.getClicks = async (req, res, next) => {
    // TODO: readHoldingRegisters
    try {
        modbus.tcp.connect(502, "192.168.250.50", { debug: null }, (err, connection) => {
            if(err) return next(err);

            connection.on("error", (err) => {
                console.log(err.message);
                if(err){
                    return next();
                }

                return JSON.stringify({msg: "Error"});
            })
            connection.readHoldingRegisters({ address: 0, quantity: 124, extra: { unitId: 0 } }, (err, data) => {

                if(data) return res.json(data);
                return JSON.stringify({msg: "Error"});
            })
        });

    } catch(ex) {
        on('error', [ex])
    }

}
`

Here is the error I am getting, this is over 3 requests:
[image]

Any help would be appreciated, thanks in advance! :)

XavegX367, 2022-12-12:
It looks like the fix to this problem was closing the connection afterwards.

Now I am getting this error after a while:

[image]

This seems to be a problem in the package itself I assume?

dresende, 2022-12-12:
Maybe. How are you closing the connection?

XavegX367, 2022-12-12:
Thanks for the response. This is how I am closing the connection:
[image]

After disabling the error in the package everything seems to be running fine.

The last error I am getting after a while:
[image]

dresende, 2022-12-12:
Can you tell me where did you disable the error so you can improve the code?

XavegX367, 2022-12-12:
> Can you tell me where did you disable the error so you can improve the code?

I disabled line 93 in lib/transport/transport.js

dresende, 2022-12-12:
That is strange because that's when it gives up trying to send a message. If you open and close connection every time this shouldn't happen. Will have to test it out.

XavegX367, 2022-12-19:
> That is strange because that's when it gives up trying to send a message. If you open and close connection every time this shouldn't happen. Will have to test it out.

Any updates on this yet?