node-modbus/stream #76: Question: Trying to reset a count function

XavegX367, 2022-12-14:
Hello, I am really new to working with Modbus and I am trying to reset the amount of times a button is pressed. the count is already working, it gets stored into 2 bytes which I retrieve with the "readHoldingRegisters" function. But when I try to create a "Reset counter" button I assume I need to use the "WriteSingleRegister" function, but I am not really sure how to send the buffer.

This is how I read the clicks:
[image]

This is the response:

[image]

The count function works by summing these 2 values after multiplying the 0 value by * 256. Value 0 and 1 need both to be set back to 0. Hopefully you are able to help.

Thanks in advance!

dresende, 2022-12-14:
Check the `README`, there's the list of methods. For WriteSingleRegister (if the device supports it), you should send:

```js
connection.writeSingleRegister({ address: 0, value: Buffer.from([ 0x01, 0x02 ]) }, (err) => {
    // ...
});
```

If it doesn't support and instead only supports multiple registers, you can use:

```js
connection.writeMultipeRegisters({ address: 0, values: [ Buffer.from([ 0x01, 0x02 ]) ] }, (err) => {
    // ...
});
```

XavegX367, 2022-12-14:
> Check the `README`, there's the list of methods. For WriteSingleRegister (if the device supports it), you should send:
>
> ```js
> connection.writeSingleRegister({ address: 0, value: Buffer.from([ 0x01, [...]

Yes I figured, but how do I get it to set both of the values I showed in the screenshot above to set it to 0?

dresende, 2022-12-14:
The rest of the functions here:

https://nodejs.org/dist/latest-v18.x/docs/api/buffer.html

XavegX367, 2022-12-14:
Thanks a lot, works like a charm!

closed 2022-12-14 by XavegX367