node-modbus/stream #78: Documentation

Venerons, 2023-03-03:
Where can I find documentation on how to use this module? Other than the readme, it doesn't seem to exists any kind of documentation around.

I'm trying to create a modbus tcp server to create a simulator, but I'm struggling to understand how to use this module.

Some things I need:

- what events can/should I listen on a server connection, other than `close`, `error` and `read-coils`?
- how can I read the actual request? The `request` object is filled with properties but there is no documentation or example on how to actually use it. Thanks to the debug I can see that the server is receiving the messages from the client, but not all seems to be catched by the `read-coils` event, and in general I don't understand how am I supposed to use the `request` object to understand the request and customize the reply.

Following my very basic server so far:

```js
// npm modules
const modbus = require('modbus-stream');

// kill logging
process.on('SIGINT', function () {
	console.error('Process killed by SIGINT');
	process.exit();
});

// exception logging
process.on('uncaughtException', function (error) {
	console.error('Uncaught Exception thrown', error);
	process.exit(1);
});

// env variables
const SERVER_PORT = process.env.SERVER_PORT || 502;

modbus.tcp.server({ debug: 'server' }, (connection) => {
	console.log('connected');
	connection.on('close', () => {
		console.log('close');
	});
	connection.on('error', () => {
		console.log('error');
	});
	connection.on('read-coils', (request, reply) => {
		console.log('read-coils');
		reply(null, [0, 0, 0, 0, 0]);
	});
}).listen(SERVER_PORT, () => {
	console.log(`Server listening on ${SERVER_PORT}`);
});
```

Venerons, 2023-03-07:
After many attempts, I managed to make a server that is working. This is the template I used (I removed all the code that actually gets the correct values)

```js
// npm modules
const modbus = require('modbus-stream');

// kill logging
process.on('SIGINT', function () {
	console.error('Process killed by SIGINT');
	process.exit();
});

// exception logging
process.on('uncaughtException', function (error) {
	console.error('Uncaught Exception thrown', error);
	process.exit(1);
});

// env variables
const SERVER_PORT = process.env.SERVER_PORT || 502;

modbus.tcp.server({ debug: 'server' }, (connection) => {
	console.log('connected');
	connection.on('close', () => {
		console.log('close');
	});
	connection.on('error', (error) => {
		console.log('error', error);
	});
	connection.on('read-coils', (request, reply) => {
		console.log('\tread-coils', request.request);
		const values = [];
		for (let i = 0; i < request.request.quantity; ++i) {
			const address = request.request.address + i;
			const value = 0; // replace this with actual value
			values.push(value);
		}
		reply(null, values);
	});
	connection.on('read-discrete-inputs', (request, reply) => {
		console.log('\tread-discrete-inputs', request.request);
	});
	connection.on('read-holding-registers', (request, reply) => {
		console.log('\tread-holding-registers', request.request);
		const values = [];
		for (let i = 0; i < request.request.quantity; ++i) {
			const address = request.request.address + i;
			const value = Buffer.from([ 0x00, 0x00 ]); // replace this with actual value
			values.push(value);
		}
		reply(null, values);
	});
	connection.on('read-input-registers', (request, reply) => {
		console.log('\tread-input-registers', request.request);
	});
	connection.on('write-single-coil', (request, reply) => {
		console.log('\twrite-single-coil', request.request);
	});
	connection.on('write-single-register', (request, reply) => {
		console.log('\twrite-single-register', request.request);
	});
	connection.on('write-multiple-coils', (request, reply) => {
		console.log('\twrite-multiple-coils', request.request);
	});
	connection.on('write-multiple-registers', (request, reply) => {
		console.log('\twrite-multiple-registers', request.request);
	});
	connection.on('data', (request, reply) => {
		console.log('\tdata', request.request);
	});
}).listen(SERVER_PORT, () => {
	console.log(`Server listening on ${SERVER_PORT}`);
});
```

stif, 2023-06-19:
It was not obvious for me as well, but i managed to write a simple modbus server script with this library.
Here is the link for the code if anybody is interested in another example how to use the library: https://github.com/nett-media/modbus-server/blob/master/modbus-server.js